ControlSystemStudio/cs-studio #2619: Cannot launch Phoebus from Eclipse

antoinetran, 2019-11-06:
Env:
CentOs 7.7
Eclipse 2019-03
CSStudio dev 4.6

When clicking in "CS-Studio" -> "Display" -> "Phoebus", in the console, I have this error:
```
java.io.IOException: Cannot run program "java" (in directory "C:\git\phoebus\phoebus-product\target"): error=2, No such file or directory
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1128)
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1071)
        at org.csstudio.phoebus.integration.PhoebusLauncherService.launchPhoebus(PhoebusLauncherService.java:56)
        at org.csstudio.phoebus.integration.PhoebusLauncherService.launch(PhoebusLauncherService.java:69)
        at org.csstudio.phoebus.integration.Launch.execute(Launch.java:18)
        at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:294)
        at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:96)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:566)
        at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
        at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:320)
        at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:254)
...
Caused by: java.io.IOException: error=2, No such file or directory
        at java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
        at java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:340)
        at java.base/java.lang.ProcessImpl.start(ProcessImpl.java:271)
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1107)
```

It refers to a hard-coded path: `C:\git\phoebus\phoebus-product\target` . This path is also in code:
https://github.com/ControlSystemStudio/cs-studio/blob/fe76c8d893176bf894dbaa85ce29ab6f38388a1f/applications/phoebus/phoebus-plugins/org.csstudio.phoebus.integration/preferences.ini

It tries to find java in this directory, but it should find it in PATH or in Eclipse vm argument.

kasemir, 2019-11-06:
> This path is also in code: .. org.csstudio.phoebus.integration/preferences.ini

It's not hardcoded, it's a preference that you can/need to configure.
If all fails, check the manual, https://phoebus-doc.readthedocs.io/en/latest/preferences.html

kasemir, 2019-11-06:
Sorry, I pointed to the manual for the new Phoebus version. For Eclipse, see here how to set preferences: http://cs-studio.sourceforge.net/docbook/ch06.html

antoinetran, 2019-11-06:
Thank you @kasemir . So it is normal that for an unconfigured CSStudio, it tries to find java in a wrong path? If yes, I will close this issue and assume it will work better when configured.

kasemir, 2019-11-06:
Each preference setting has some default. The current default is what Kunal selected, presumably because it worked well when he developed that section of code. I don't see a way to predict where people would install phoebus, so this really needs to be a preference setting, everybody needs to configure it, there is no universal default setting except maybe "/path/to/your/phoebus-product" so that the default will for sure fail for everybody, making it even more obvious that it needs to be configured.

antoinetran, 2019-11-06:
While I agree with what you said, I still find it weird that it tries to launch java from the phoebus directory, instead of either $PATH or the same used to launch Eclipse. But that's fine for me, closing the issue and thank you. I will reopen it if the issue (java not found) persists after configuration.

closed 2019-11-06 by antoinetran

kasemir, 2019-11-06:
Maybe there is a way to have the default preference be empty, and handle an empty setting to mean "use $PATH", but Kunal @shroffk would know best if that's feasible. The default setting with `phoebus_version=0.0.1-SNAPSHOT` would also need to be updated since we're now on 4.6.0.

shroffk, 2019-11-06:
The default path was previously set to the same directory as the phoebus installation cause the eclipse cs-studio was still using java8 and phoebus needed java 11.

Now that both the eclipse and cs-studio are based on jdk11 we can remove this preference completely and use the java available on path.