ControlSystemStudio/cs-studio #2623: Scrollbar Steals Focus from Overlayed TextInput

dylan171, 2019-12-20:
I have an OPI with a TextInput control overlayed on a Scrollbar control.

[image]

In CS Studio 4.5, clicking on the TextInput would cause the text field to have focus and allow the user to input a new value. 

In CS Studio 4.6, clicking on TextInput activates a pageup/pagedown event on the scrollbar, and the text field do not get focus.

Example OPI file: [Scrollbar_Issue_CSS46.opi.zip](https://github.com/ControlSystemStudio/cs-studio/files/3989001/Scrollbar_Issue_CSS46.opi.zip)

dylan171, 2019-12-20:
I have tested on both Linux (GTK3) and Windows with the same result.

kasemir, 2019-12-20:
For what it's worth, this would for now be OK in the Display Builder running on Phoebus:

[image]

But I don't want to guarantee that it'll always function that way, and there is a dedicated 'Spinner' widget, shown at the bottom, for the purpose of having both a text entry and up/down buttons.

In BOY, the reason for the somewhat unpredictably behavior could be that BOY uses a mix of SWT and Draw2D widgets, and who gets a mouse click event when they're layered isn't always clear. I think at one time there was an issue that one type of widget would always show up on top, no matter how you think they should be layered, because again these are different widget technologies.

So if you want to guarantee this behavior in BOY, you might have to implement a 'Spinner' type widget.

kasemir, 2019-12-20:
I think BOY actually has a spinner widget, so how about using that?

dylan171, 2019-12-20:
We will try to use the spinner widget instead.

For Phoebus, maybe a horizontal option could be considered:

[image]

Personally I find the buttons on the regular spinner widget too small.