ControlSystemStudio/cs-studio #2625: Show All/Hide All action in Databrowser

ifmalcuaz, 2020-01-27:
On the databrowser if you wanted to show or hide all the traces you need to individually click on the "Show" check boxes for each PV. It would be useful to have a Show/Hide All button somewhere; one suggestion is to just add two context menu actions when right clicking on the plot (Show All, Hide All, or similar):

[image]

cc @willrogers

willrogers, 2020-01-27:
@shroffk this was a user request

kasemir, 2020-01-27:
Some of our users are in the habit of adding more and more items to a plot, then hide most of them because they're not really necessary, ..., and then complain that things slow down. Would a 'remove all' be more useful?

shroffk, 2020-01-27:
Hmm... I have similar usage, where people add many traces (sometimes raw)... then hide most of them then and then wonder why databrowser uses so much memory.

> Would a 'remove all' be more useful?

If I understand the use case correctly, people still want to keep the traces in the properties but just not show/plot them.

I think we can add the show all/hide all actions since I don't think they make things worse...

In the future, when a trace is hidden we could have databrowser discard the retrieved archive data and pause the pv connection.

Gir4ffe, 2020-01-27:
@willrogers asked me to elaborate on this. 

Currently we have some plt's setup that let us view an entire system. If I use our cooling water system as an example, there are many pv's over 4 chillers and associated equipment. 

When fault finding, it's really handy to be able to see everything together, to look for patterns/coincidences, also really handy to single one pv/ a few out. 

Having a select all would mean we could drill down and build up from one pv. Currently to do this we have to copy the pv's we want to a new plt and re-set the axis.

willrogers, 2020-01-29:
Presumably, if I understand correctly, you could scroll through all the PVs and untick them rather than create a whole new plot? Not that I'm saying this is a good solution particularly.

Gir4ffe, 2020-01-29:
You can indeed scroll through and tick/untick what you need. 

I have attached a screenshoot of pvs for the plt of 1 of our 4 chillers. As you can see, ticking and un-ticking takes a while :)  

[image]

willrogers, 2020-01-29:
Thanks, I think that illustrates the problem.

shroffk, 2020-01-29:
Thanks @Gir4ffe, your example clearly identifies your use case. 

Below I have described the workflow I have showed my users... 

I am guessing you are aware of the "edit selected items" 
[image]

"show"
[image]

So it comes down to

Ctrl + a> "edit selected items" > check "show" > OK

So there is one extra step here compared to what you requested, but it is surely better than individually checking :)

kasemir, 2020-01-29:
I'll add that "Edit Selected Items" support to the phoebus version, it hadn't been imported, yet.

Gir4ffe, 2020-01-30:
@shroffk I was sort of aware - not had my official CSS training just yet (though I have had some)

Just had a go - works for me. 

Thanks very much @willrogers @shroffk @kasemir

willrogers, 2020-01-30:
OK, let's close this one then.

closed 2020-01-30 by willrogers