ControlSystemStudio/cs-studio #2628: Default pointer in Databrowser

ifmalcuaz, 2020-01-29:
A user has requested that the default mouse pointer in the databrowser to be the plain mouse pointer instead of pan plot. We can implement a dropdown in the Preferences->Trends->Data Browser to set the default pointer or just change it directly so it's always set to plain mouse by default.

cc @willrogers

willrogers, 2020-01-29:
Thinking about it, this would also help with another problem we see: a single click when 'pan plot' is selected makes the plot dirty. This doesn't happen if 'plain mouse pointer' is the default.

kasemir, 2020-01-29:
The default mouse pointer mode is plain pointer in the JavaFX-based data browser and the one in the Phoebus-based CS-Studio, so makes sense to also pick that for the SWT-based version.

closed 2020-02-13 by willrogers