ControlSystemStudio/cs-studio #2631: Product Materialization Build Error

dylan171, 2020-01-31:
My CS Studio build was working... the last time I built everything successfully was Jan 15, 2020.

Yesterday, I tried again and it failed during the final step, product materialization:
```
[INFO] --- tycho-p2-repository-plugin:1.3.0:assemble-repository (default-assemble-repository) @ frib-repository ---
[WARNING] Mirror tool: Problems resolving provisioning plan.: [Unable to satisfy dependency from cs-studio-frib 4.6.3 to org.eclipse.equinox.p2.iu; org.eclipse.equinox.executable.feature.group [3.8.600.v20191014-2025,3.8.600.v20191014-2025].; Unable to satisfy dependency from cs-studio-frib 4.6.3 to org.eclipse.equinox.p2.iu; org.eclipse.equinox.p2.core.feature.feature.group [1.6.400.v20191120-0247,1.6.400.v20191120-0247].; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.equinox.event 4.6.3 to osgi.bundle; org.eclipse.equinox.event 1.5.300.v20191001-1333.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.equinox.event 4.6.3 to osgi.bundle; org.eclipse.equinox.event 1.5.300.v20191001-1333.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.equinox.p2.reconciler.dropins 4.6.3 to osgi.bundle; org.eclipse.equinox.p2.reconciler.dropins 1.3.200.v20191016-0510.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.equinox.p2.reconciler.dropins 4.6.3 to osgi.bundle; org.eclipse.equinox.p2.reconciler.dropins 1.3.200.v20191016-0510.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.equinox.common 4.6.3 to osgi.bundle; org.eclipse.equinox.common 3.10.600.v20191004-1420.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.equinox.common 4.6.3 to osgi.bundle; org.eclipse.equinox.common 3.10.600.v20191004-1420.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.apache.felix.scr 4.6.3 to osgi.bundle; org.apache.felix.scr 2.1.14.v20190123-1619.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.apache.felix.scr 4.6.3 to osgi.bundle; org.apache.felix.scr 2.1.14.v20190123-1619.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.core.runtime 4.6.3 to osgi.bundle; org.eclipse.core.runtime 3.17.0.v20191122-2104.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.core.runtime 4.6.3 to osgi.bundle; org.eclipse.core.runtime 3.17.0.v20191122-2104.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.osgi 4.6.3 to osgi.bundle; org.eclipse.osgi 3.15.100.v20191114-1701.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.osgi 4.6.3 to osgi.bundle; org.eclipse.osgi 3.15.100.v20191114-1701.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.equinox.simpleconfigurator 4.6.3 to osgi.bundle; org.eclipse.equinox.simpleconfigurator 1.3.400.v20191015-1836.; Unable to satisfy dependency from toolinggtk.linux.x86_64org.eclipse.equinox.simpleconfigurator 4.6.3 to osgi.bundle; org.eclipse.equinox.simpleconfigurator 1.3.400.v20191015-1836.; Unable to satisfy dependency from toolingorg.eclipse.equinox.launcher 1.5.600.v20191014-2022 to osgi.bundle; org.eclipse.equinox.launcher 1.5.600.v20191014-2022.; Unable to satisfy dependency from toolingorg.eclipse.equinox.launcher 1.5.600.v20191014-2022 to osgi.bundle; org.eclipse.equinox.launcher 1.5.600.v20191014-2022.]
[WARNING] More information on the preceding warning(s) can be found here:
[WARNING] - http://wiki.eclipse.org/Tycho_Messages_Explained#Mirror_tool
[WARNING] Mirror tool: Problems resolving provisioning plan.: [Unable to satisfy dependency from cs-studio-frib 4.6.3 to org.eclipse.equinox.p2.iu; org.eclipse.equinox.executable.feature.group [3.8.600.v20191014-2025,3.8.600.v20191014-2025].; Unable to satisfy dependency from cs-studio-frib 4.6.3 to org.eclipse.equinox.p2.iu; org.eclipse.equinox.p2.core.feature.feature.group [1.6.400.v20191120-0247,1.6.400.v20191120-0247].; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.osgi 4.6.3 to osgi.bundle; org.eclipse.osgi 3.15.100.v20191114-1701.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.osgi 4.6.3 to osgi.bundle; org.eclipse.osgi 3.15.100.v20191114-1701.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.equinox.event 4.6.3 to osgi.bundle; org.eclipse.equinox.event 1.5.300.v20191001-1333.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.equinox.event 4.6.3 to osgi.bundle; org.eclipse.equinox.event 1.5.300.v20191001-1333.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.equinox.simpleconfigurator 4.6.3 to osgi.bundle; org.eclipse.equinox.simpleconfigurator 1.3.400.v20191015-1836.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.equinox.simpleconfigurator 4.6.3 to osgi.bundle; org.eclipse.equinox.simpleconfigurator 1.3.400.v20191015-1836.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.apache.felix.scr 4.6.3 to osgi.bundle; org.apache.felix.scr 2.1.14.v20190123-1619.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.apache.felix.scr 4.6.3 to osgi.bundle; org.apache.felix.scr 2.1.14.v20190123-1619.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.core.runtime 4.6.3 to osgi.bundle; org.eclipse.core.runtime 3.17.0.v20191122-2104.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.core.runtime 4.6.3 to osgi.bundle; org.eclipse.core.runtime 3.17.0.v20191122-2104.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.equinox.common 4.6.3 to osgi.bundle; org.eclipse.equinox.common 3.10.600.v20191004-1420.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.equinox.common 4.6.3 to osgi.bundle; org.eclipse.equinox.common 3.10.600.v20191004-1420.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.equinox.p2.reconciler.dropins 4.6.3 to osgi.bundle; org.eclipse.equinox.p2.reconciler.dropins 1.3.200.v20191016-0510.; Unable to satisfy dependency from toolingwin32.win32.x86_64org.eclipse.equinox.p2.reconciler.dropins 4.6.3 to osgi.bundle; org.eclipse.equinox.p2.reconciler.dropins 1.3.200.v20191016-0510.; Unable to satisfy dependency from toolingorg.eclipse.equinox.launcher.win32.win32.x86_64 1.1.1100.v20190907-0426 to osgi.bundle; org.eclipse.equinox.launcher.win32.win32.x86_64 1.1.1100.v20190907-0426.; Unable to satisfy dependency from toolingorg.eclipse.equinox.launcher.win32.win32.x86_64 1.1.1100.v20190907-0426 to osgi.bundle; org.eclipse.equinox.launcher.win32.win32.x86_64 1.1.1100.v20190907-0426.; Unable to satisfy dependency from toolingorg.eclipse.equinox.launcher 1.5.600.v20191014-2022 to osgi.bundle; org.eclipse.equinox.launcher 1.5.600.v20191014-2022.; Unable to satisfy dependency from toolingorg.eclipse.equinox.launcher 1.5.600.v20191014-2022 to osgi.bundle; org.eclipse.equinox.launcher 1.5.600.v20191014-2022.]
[WARNING] More information on the preceding warning(s) can be found here:
[WARNING] - http://wiki.eclipse.org/Tycho_Messages_Explained#Mirror_tool
[WARNING] Mirror tool: Problems resolving provisioning plan.: [Unable to satisfy dependency from cs-studio-frib 4.6.3 to org.eclipse.equinox.p2.iu; org.eclipse.equinox.executable.feature.group [3.8.600.v20191014-2025,3.8.600.v20191014-2025].; Unable to satisfy dependency from cs-studio-frib 4.6.3 to org.eclipse.equinox.p2.iu; org.eclipse.equinox.p2.core.feature.feature.group [1.6.400.v20191120-0247,1.6.400.v20191120-0247].; Unable to satisfy dependency from toolingcocoa.macosx.x86_64org.eclipse.osgi 4.6.3 to osgi.bundle; org.eclipse.osgi 3.15.100.v20191114-1701.; Unable to satisfy dependency from toolingcocoa.macosx.x86_64org.eclipse.osgi 4.6.3 to osgi.bundle; org.eclipse.osgi 3.15.100.v20191114-1701.; Unable to satisfy dependency from toolingcocoa.macosx.x86_64org.eclipse.equinox.p2.reconciler.dropins 4.6.3 to osgi.bundle; org.eclipse.equinox.p2.reconciler.dropins 1.3.200.v20191016-0510.; Unable to satisfy dependency from toolingcocoa.macosx.x86_64org.eclipse.equinox.p2.reconciler.dropins 4.6.3 to osgi.bundle; org.eclipse.equinox.p2.reconciler.dropins 1.3.200.v20191016-0510.; Unable to satisfy dependency from toolingcocoa.macosx.x86_64org.eclipse.equinox.common 4.6.3 to osgi.bundle; org.eclipse.equinox.common 3.10.600.v20191004-1420.; Unable to satisfy dependency from toolingcocoa.macosx.x86_64org.eclipse.equinox.common 4.6.3 to osgi.bundle; org.eclipse.equinox.common 3.10.600.v20191004-1420.; Unable to satisfy dependency from toolingcocoa.macos [... cut ...]
An old bug [report](https://bugs.eclipse.org/bugs/show_bug.cgi?id=477089) suggested that I'm missing the org.eclipse.equinox.p2.extras.feature from my product. But I double checked and it is included in my product.

Instead of a missing feature/bundle, it appears that product materialization is looking for a newer version then I have in the local target repository. For example:
```
Unable to satisfy dependency from cs-studio-frib 4.6.3 to org.eclipse.equinox.p2.iu; org.eclipse.equinox.p2.core.feature.feature.group [1.6.400.v20191120-0247,1.6.400.v20191120-0247]
```
But the feature I have is org.eclipse.equinox.p2.core.feature_1.6.0.v20181123-1405.jar

Any suggestions would be appreciated.

shroffk, 2020-07-29:
hmm... are you able to build the common product?

dxmaxwell, 2020-07-29:
I never tried building the common product.

It seems like my build was somehow finding more recent version of certain libraries, but I was able to "fix" this issue by adding the following filter configuration to the pom.xml
```
         <filters>
            <filter>
              <type>p2-installable-unit</type>
              <id>org.eclipse.equinox.p2.core.feature.feature.group</id>
              <restrictTo>
                 <version>1.6.0.v20181123-1405</version>
              </restrictTo>
            </filter>
            <filter>
              <type>eclipse-plugin</type>
              <id>org.apache.felix.scr</id>
              <restrictTo>
                 <version>2.0.14.v20180822-1822</version>
              </restrictTo>
            </filter>
            <filter>
              <type>eclipse-plugin</type>
              <id>org.eclipse.core.runtime</id>
              <restrictTo>
                 <version>3.15.100.v20181107-1343</version>
              </restrictTo>
            </filter>
            <filter>
              <type>eclipse-plugin</type>
              <id>org.eclipse.equinox.common</id>
              <restrictTo>
                 <version>3.10.200.v20181021-1645</version>
              </restrictTo>
            </filter>
            <filter>
              <type>eclipse-plugin</type>
              <id>org.eclipse.equinox.event</id>
              <restrictTo>
                 <version>1.5.0.v20181008-1938</version>
              </restrictTo>
            </filter>
            <filter>
              <type>eclipse-plugin</type>
              <id>org.eclipse.equinox.p2.reconciler.dropins</id>
              <restrictTo>
                 <version>1.3.0.v20181023-0914</version>
              </restrictTo>
            </filter>
            <filter>
              <type>eclipse-plugin</type>
              <id>org.eclipse.equinox.simpleconfigurator</id>
              <restrictTo>
                 <version>1.3.200.v20181101-1115</version>
              </restrictTo>
            </filter>
            <filter>
              <type>eclipse-plugin</type>
              <id>org.eclipse.osgi</id>
              <restrictTo>
                 <version>3.13.200.v20181130-2106</version>
              </restrictTo>
            </filter>
            <filter>
              <type>eclipse-plugin</type>
              <id>org.eclipse.equinox.launcher</id>
              <restrictTo>
                 <version>1.5.200.v20180922-1751</version>
              </restrictTo>
            </filter>
            <filter>
              <type>eclipse-plugin</type>
              <id>org.eclipse.equinox.launcher.win32.win32.x86_64</id>
              <restrictTo>
                 <version>1.1.900.v20180922-1751</version>
              </restrictTo>
            </filter>
          </filters>
```
I was stumped by the error for a while, but eventually I figured this out. Since it seems no one else encountered it, I will close this issue.

closed 2020-07-29 by dxmaxwell