ControlSystemStudio/cs-studio #2638: Create Log Entry Dialog Failure

dylan171, 2020-02-13:
When attempting to create a log entry with an attachment, the following error occurs:
[image]
[image]

The stack trace is as follows:
```
javax.ws.rs.WebApplicationException: javax.xml.bind.JAXBException: Implementation of JAXB-API has not been found on module path or classpath.
 - with linked exception:
[java.lang.ClassNotFoundException: com.sun.xml.bind.v2.ContextFactory]
	at com.sun.jersey.core.provider.jaxb.AbstractRootElementProvider.readFrom(AbstractRootElementProvider.java:115)
	at com.sun.jersey.api.client.ClientResponse.getEntity(ClientResponse.java:634)
	at com.sun.jersey.api.client.ClientResponse.getEntity(ClientResponse.java:586)
	at com.sun.jersey.api.client.WebResource.handle(WebResource.java:686)
	at com.sun.jersey.api.client.WebResource.access$200(WebResource.java:74)
	at com.sun.jersey.api.client.WebResource$Builder.post(WebResource.java:570)
	at edu.msu.nscl.olog.api.OlogClientImpl.add(OlogClientImpl.java:831)
	at org.csstudio.logbook.olog.OlogLogbookClient.addAttachment(OlogLogbookClient.java:203)
	at org.csstudio.logbook.olog.OlogLogbookClient.createLogEntry(OlogLogbookClient.java:162)
	at org.csstudio.logbook.ui.LogEntryBuilderDialog$1.run(LogEntryBuilderDialog.java:144)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
Caused by: javax.xml.bind.JAXBException: Implementation of JAXB-API has not been found on module path or classpath.
 - with linked exception:
[java.lang.ClassNotFoundException: com.sun.xml.bind.v2.ContextFactory]
	at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:278)
	at javax.xml.bind.ContextFinder.find(ContextFinder.java:397)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:721)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:662)
	at com.sun.jersey.core.provider.jaxb.AbstractJAXBProvider.getStoredJAXBContext(AbstractJAXBProvider.java:196)
	at com.sun.jersey.core.provider.jaxb.AbstractJAXBProvider.getJAXBContext(AbstractJAXBProvider.java:188)
	at com.sun.jersey.core.provider.jaxb.AbstractJAXBProvider.getUnmarshaller(AbstractJAXBProvider.java:140)
	at com.sun.jersey.core.provider.jaxb.AbstractJAXBProvider.getUnmarshaller(AbstractJAXBProvider.java:123)
	at com.sun.jersey.core.provider.jaxb.AbstractRootElementProvider.readFrom(AbstractRootElementProvider.java:111)
	... 10 more
Caused by: java.lang.ClassNotFoundException: com.sun.xml.bind.v2.ContextFactory
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:583)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	at org.eclipse.osgi.internal.framework.ContextFinder.loadClass(ContextFinder.java:135)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	at javax.xml.bind.ServiceLoaderUtil.nullSafeLoadClass(ServiceLoaderUtil.java:122)
	at javax.xml.bind.ServiceLoaderUtil.safeLoadClass(ServiceLoaderUtil.java:155)
	at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:276)
	... 18 
```

dylan171, 2020-02-13:
Problem seems to be an oversight in the OLog client implementation that did not do the properly class loader wrangling when making a request to add attachments to a log entry.

See PR [80](https://github.com/epics-rip/cs-studio-thirdparty/pull/80) in the CS Studio Thirdparty repo for details.

shroffk, 2020-07-28:
Thanks for fixing this

shroffk, 2020-07-28:
When I tested this we were able to successfully make log entries despite the error

dxmaxwell, 2020-07-29:
Yes, if I remember correctly the log entry is successfully created in olog, but the intended attachments were missing.

shroffk, 2020-07-29:
Oh, in the test I ran even the attachments were included....strange.
In any case could you approve the PR

closed 2020-07-29 by shroffk