ControlSystemStudio/cs-studio #2639: LED font color change works only for the first time

ihtao127, 2020-04-15:
I realy like the feature that the LED font changes the color base on the backgroud color, if the color is dark, the font changes to a white color, it is good for the readibility.

But it only works once when the OPI is initialized by running it or reload it.

Please look at [the video](https://youtu.be/qpW8GVofQXk).

kasemir, 2020-04-15:
Since this is for the display builder and phoebus, you could have posted this issue there, not in the Eclipse/BOY based repo. But anyway, the display builder examples for "LED" contain a more reproducible setup since it's using local PVs, so anybody can execute that display. When I try it, the color changes as expected:

[image]
[image]

Same with the "Byte Monitor" example:

[image]

What do you get for those examples?

ihtao127, 2020-04-15:
Thank you very much for your answer and please forgive me to post this issue in the wrong repo.

I've had a look at the examples, and now I found the probable reason, maybe it is because my Enum PV doesn't have the right definition on the fields, which means that the bi record just have a "ONAM" filed and there's no "ZNAM" field.

I will fix the field definition and test this again.

Thank you again.

ihtao127, 2020-04-16:
I have created a new issue in the **_[Phoebus repo](https://github.com/ControlSystemStudio/phoebus/issues/1269)_**.

closed 2020-04-16 by ihtao127