ControlSystemStudio/cs-studio #2640: Another Batik dependency problem

willrogers, 2020-04-15:
I have built the latest master branch. I then run the following command:

```
./cs-studio -nosplash -application org.csstudio.opibuilder.converter.edl -h
```
I get a little pop-up saying 'An error has occurred. Please see the log file ...

I have attached the log file. Unfortunately it is another batik depdendency error and I cannot figure it out. Any ideas @shroffk?

I first saw this when re-enabling the unit tests on our Jenkins build.

[1586968616661.log](https://github.com/ControlSystemStudio/cs-studio/files/4482521/1586968616661.log)

shroffk, 2020-04-25:
Hmm... No I have not come across this one as yet.
I don't use the converter much which is why we might have missed it.

willrogers, 2020-04-27:
I think all you need to do to find this error is run the unit tests.

willrogers, 2021-08-10:
I think this was resolved by #2647 .

closed 2021-08-10 by willrogers