ControlSystemStudio/cs-studio #2645: gtk2 no longer supported.

shroffk, 2020-07-28:
https://www.eclipse.org/eclipse/news/4.10/platform.php#gtk2-removal

Dropped support for GTK2 |
GTK2 is no longer a supported target environment for Eclipse/SWT on Linux in 4.10, and Eclipse/SWT applications will no longer run on GTK2. Furthermore the SWT_GTK3 environment variable is ignored, as only GTK3 is supported now.
-- | --

However, with gtk3 there are issues on debian.

willrogers, 2020-08-05:
Do you know what the issues are? As far as I can tell GTK3 works fine on RHEL7 and does also fix a few niggles from GTK2.

shroffk, 2020-09-02:
https://bugs.eclipse.org/bugs/show_bug.cgi?id=517671
https://bugs.eclipse.org/bugs/show_bug.cgi?id=528683

These issues describe the same/similar problem.

willrogers, 2020-09-03:
Presumably you have tried these two workarounds:

```
By creating a new user, copying everything within ~/.config from my usual user to the new user, and then iteratively removing config files, I have found the culprit on my system:

  GTK_IM_MODULE=xim

I had set this environment variable so that XCompose would work in Gtk+ applications. (The default Gtk+ input method was ignoring my ~/.XCompose file.)

If I unset GTK_IM_MODULE before starting Eclipse, the flickering problem goes away. Surprisingly, my custom compose sequences still work in Eclipse, so I guess Gtk+3 has added support for XCompose in its default input method module.

Would be great to know *why* XIM causes flicker in Eclipse, but I guess that's an exercise for the SWT developers. :)
```

and 

```
Hi, 

Just my two little cents for those who are looking for a temporary fix. Flickering disappeared by removing gtk3 setting.ini in /home/$USER/.config/gtk3/ (don't forget to login/out afterward).
```

shroffk, 2020-09-03:
yes, tried these two

willrogers, 2020-09-03:
I can't see any discussion in these tickets about gdm or lightdm.