ControlSystemStudio/cs-studio #2694: Phoebus display option in CS-Studio

rjwills28, 2021-06-15:
There is an option in CS-Studio to launch the Phoebus application from the top toolbar > `CS-Studio` > `Display` > `Phoebus`. This only works if the `phoebus_home` and `phoebus_version` have been set in  `applications/phoebus/phoebus-plugins/org.csstudio.phoebus.integration/preferences.ini`. Otherwise exceptions are thrown in the log and nothing happens. 
[image]

Do you think there should be a preference to define whether or not the Phoebus launch is shown in the `CS-Studio` > `Display` menu in the case that Phoebus has not been installed?

shroffk, 2021-07-06:
Not including the Phoebus integration plugin/feature in your product would achieve the same result.
I don't like preferences to disable a feature which is included without intention to use...

kasemir, 2021-07-06:
Yes, simply not including what you don't need is most straight forward.
You can update your site-specific product build to only include what you need in the first place.
After the fact, I also find it practical to simply delete the corresponding *.jar file from the product.

willrogers, 2021-09-23:
After looking into this, I find that opibuilder depends on `org.csstudio.phoebus.integration`. So anyone who includes opibuilder gets the Phoebus menu entry, or so it seems. We have not changed the preference `use_phoebus=false`.

shroffk, 2021-09-23:
I will look into this.
The dependency might have happened with the "open in Phoebus" action