ControlSystemStudio/cs-studio #2704: Update cs-studio version

willrogers, 2021-08-10:
On the master branch in pom.xml we still have

```
  <artifactId>cs-studio</artifactId>
  <version>4.6.0-SNAPSHOT</version>
```

This needs to be updated to 4.7.0-SNAPSHOT. How is this done @shroffk? Is this part of the Maven release process?

cc @aawdls @mjgaughran

shroffk, 2021-08-10:
Yes, 
All the snapshot and setting on the next development version is part to the mvn release plugin.
I am not sure if Tycho complicates things.

I don't think the 4.6 release branch and the 4.7.x release branch still used 4.6.0-SNAPSHOT.

So here is what I think we should do
1. In the future make releases off the master branch itself...I think managing release branches makes sense if we have many patches but looking at the past usage I think this is us unnecessary complexity.
2. perform a release of 4.6
3. prepare for 4.7x development

willrogers, 2021-08-10:
I don't think anyone actually wants the 4.6 release. But if that's the way we get to the 4.7 version that's fine by me.

How do we actually do it?