ControlSystemStudio/cs-studio #2717: cs-studio crashing on first start

codedump, 2022-08-24:
Hello,

I'm attempting to start cs-studio on a fresh Fedora Silverblue toolbox (think of it as of a newly installed Fedora Linux system). Essentially, the steps I do after download and unpacking are these:
```
$ dnf install java-openjdk-latest
$ ./cs-studio
```

It crashes and leaves behind the following log file contents in `~/.cs-studio/.metadata/.log`:
```
SESSION 2022-08-24 15:31:59.299 -----------------------------------------------                                                                                                     
eclipse.buildId=unknown                                                                                                                                                              
java.version=18.0.2                                                                                                                                                                  
java.vendor=Red Hat, Inc.                                                                                                                                                            
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en                                                                                                                           
Command-line arguments:  -os linux -ws gtk -arch x86_64                                                                                                                              
                                                                                                                                                                                     
!ENTRY org.eclipse.equinox.ds 1 0 2022-08-24 15:32:00.000                                                                                                                            
!MESSAGE Could not bind a reference of component org.csstudio.diirt.util.RegisterDatasource. The reference is: Reference[name = DataSourceProvider, interface = org.diirt.datasource.
DataSourceProvider, policy = dynamic, cardinality = 0..n, target = null, bind = registerDatasource, unbind = deregisterDatasource]                                                   
                                                                                                                                                                                     
!ENTRY org.eclipse.equinox.ds 1 0 2022-08-24 15:32:00.001                                                                                                                            
!MESSAGE Could not bind a reference of component org.csstudio.diirt.util.RegisterFormulaFunctionSet. The reference is: Reference[name = FormulaFunctionSet, interface = org.diirt.dat
asource.formula.FormulaFunctionSet, policy = dynamic, cardinality = 0..n, target = null, bind = registerFormulaFunctionSet, unbind = deregisterFormulaFunctionSet]                   
                                                                                                                                                                                     
!ENTRY org.eclipse.equinox.ds 1 0 2022-08-24 15:32:00.001                                                                                                                            
!MESSAGE Could not bind a reference of component org.csstudio.diirt.util.RegisterService. The reference is: Reference[name = ServiceProvider, interface = org.diirt.service.ServicePr
ovider, policy = static, cardinality = 0..n, target = null, bind = registerService, unbind = null]                                                                                   
                                                                                                                                                                                     
!ENTRY org.eclipse.osgi 4 0 2022-08-24 15:32:06.773                                                                                                                                  
!MESSAGE Application error                                                                                                                                                           
!STACK 1                                                                                                                                                                             
org.eclipse.e4.core.di.InjectionException: java.lang.NoClassDefFoundError: javax/annotation/PostConstruct                                                                            
        at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:386)                                                                                          
        at org.eclipse.e4.core.internal.di.InjectorImpl.make(InjectorImpl.java:294)
        at org.eclipse.e4.core.contexts.ContextInjectionFactory.make(ContextInjectionFactory.java:162)
        at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createDefaultHeadlessContext(E4Application.java:490)
        at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createDefaultContext(E4Application.java:504)
        at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createE4Workbench(E4Application.java:203)
        at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:632)
        at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
        at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:610)
        at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
        at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:99)
        at org.csstudio.startup.application.Application.startApplication(Application.java:265)
        at org.csstudio.startup.application.Application.start(Application.java:119)
        at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
        at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:104)
        at java.base/java.lang.reflect.Method.invoke(Method.java:577)
        at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
        at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
        at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
        at org.eclipse.equinox.launcher.Main.main(Main.java:1492)
Caused by: java.lang.NoClassDefFoundError: javax/annotation/PostConstruct
        at org.eclipse.e4.core.internal.di.InjectorImpl.inject(InjectorImpl.java:151)
        at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:375)
        ... 23 more
Caused by: java.lang.ClassNotFoundException: javax.annotation.PostConstruct cannot be found by org.eclipse.e4.core.di_1.6.1.v20160712-0927
        at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:410)
        at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:372)
        at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:364)
        at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:161)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
        ... 25 more

!ENTRY org.eclipse.e4.ui.workbench 4 0 2022-08-24 15:32:06.777
[... 15 lines cut ...]
        at org.eclipse.osgi.internal.framework.EquinoxEventPublisher.publishBundleEvent(EquinoxEventPublisher.java:120)
        at org.eclipse.osgi.internal.framework.EquinoxEventPublisher.publishBundleEvent(EquinoxEventPublisher.java:112)
        at org.eclipse.osgi.internal.framework.EquinoxContainerAdaptor.publishModuleEvent(EquinoxContainerAdaptor.java:156)
        at org.eclipse.osgi.container.Module.publishEvent(Module.java:476)
        at org.eclipse.osgi.container.Module.doStop(Module.java:634)
        at org.eclipse.osgi.container.Module.stop(Module.java:498)
        at org.eclipse.osgi.container.SystemModule.stop(SystemModule.java:202)
[... cut ...]
        at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:372)
        at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:364)
        at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:161)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
        ... 21 more

!ENTRY org.eclipse.core.resources 2 10035 2022-08-24 15:32:06.792
!MESSAGE The workspace will exit with unsaved changes in this session.
```

What is going wrong?

Thanks & Cheers,
F.

shroffk, 2022-09-20:
CS-Studio is not supported for JDK18
You need to use jdk8 or jdk11

closed 2022-09-20 by shroffk

kasemir, 2022-09-20:
Just for the record: Is JDK8 really still supported?
Does the JDK need to be 64 bit, or is JDK 8 on 32 bit still supported?

shroffk, 2022-09-20:
I guess I should have only mentioned jdk11.
I am using jdk11, I have not tested with jdk8 in awhile now. I think ITER might be still using it with jdk8.

Even with jdk11 I remember some issues associated with missing/deprecated modules.

@codedump we would recommend using https://github.com/ControlSystemStudio/phoebus/ if you can

codedump, 2022-09-20:
> CS-Studio is not supported for JDK18 You need to use jdk8 or jdk11

@shroffk, may I suggest documenting this? ;-) Or maybe it's already documented but I couldn't find it (?).

I came through the website, downloaded the package didn't find any README or hint on the download site itself, and had a look at what I assume is [closest to the documentation's "installation" section](http://cs-studio.sourceforge.net/docbook/ch04.html).

shroffk, 2022-09-20:
I agree... for the time being I added a blurb on the readme.
https://github.com/ControlSystemStudio/cs-studio/pull/2718

Have you considered moving to Phoebus

kasemir, 2022-09-20:
If you're just starting to look at this, didn't even get it to run, you'll definitely be better off starting with https://github.com/ControlSystemStudio/phoebus/ which no longer requires Eclipse, builds faster, and runs with JDK 11-18

codedump, 2022-09-20:
@shroffk @kasemir Thank you for pointing me towards Phoebus, I'll definitely have a look at it! :-)

I didn't (still don't) know the difference between Phoebus and CS-Studio. I was about to evaluate tooling around EPICS, starting with CS-Studio, when I hit the problem I described. Meanwhile I was also nudged towards Phoebus by people at BESSY (Berlin), so I'll definitely have a closer look at it.

I'm also thinking about adaptations of tooling from other areas (e.g. [Prometheus monitoring](https://prometheus.io) and [Grafana](https://grafana.com)) for EPICS dashboarding, although this is a one-way communication only -- just monitoring, no controlling. But the advantage is that the user and developer base is *huge*. If one would write e.g. a Prometheus exporter for EPICS, one would instantly gain access to a really vast, well-documented infrastructure. But of course, one would leave behind the goodies that CS-Studio / Phoebus has to offer that go beyond monitoring...

kasemir, 2022-09-20:
CS-Studio was first built on Eclipse to use its SWT graphics, preference system, plugin .. and built system.
Eventually compiling CSS took 45 minutes, if it built at all.

So we started to look for alternatives and found JavaFX as a new graphics lib, plain Java preferences as very similar to the Eclipse preferences, plain Java SPI as very similar to Eclipse (sorry: OSGi) plugins, and we can use maven or ant to build.
So "phoebus" has now pretty much all the functionality, it is the "current" version of CS-Studio, builds in 5..10 minutes. It's where new stuff is added (PVAccess on IPv6, alarm system that runs with Kafka instead of RDB+JMS, ..)