ControlSystemStudio/cs-studio #2728: How to reduce the wait time of CSS connection when IOC start?

1458861693, 2024-04-17:
Hi，I am using CS-Studio 4.6.4-SNAPSHOT. I want to know how to reduce the wait time of CSS connection to IOC when IOC first start or reboot? I feel the wait time of cs-studio connect to IOC is a little long. Thanks for your help.

1458861693, 2024-04-18:
I don't find "cash" command on my computer or IOC Shell. Is a epics channel access command line tools? Maybe the Version of my epics base don't contain this command, the Version is EPICS R7.0.8. 

Can I decrease the connection time between CSS and IOC server by reduce the value of "EPICS_CAS_BEACON_PERIOD"? The default value of this parameter in CS-Studio is 30s.

kasemir, 2024-04-18:
Spell checker...  It's `casw`, CA server watcher.

Changing the EPICS_CAS_BEACON_PERIOD is a bad idea.
You'd have to do that for all servers and clients on your network.
As you reduce the beacon period, your servers may not be able to keep up with the exact beacon period. Clients will then erroneously wake up, thinking there's a new server online, re-issue their search requests, and then you wonder why your network is flooded with UDP messages from the search requests. Ask me how I know. 
Also, JCA doesn't necessarily read EPICS_CAS_BEACON_PERIOD.

Instead of trying to change the beacon behavior, first check if your clients see the beacons at all (casw, camonitor).

If you do want to tweak the search behavior of the CA client, check the system properties introduced in https://github.com/epics-base/jca/pull/53