ControlSystemStudio/cs-studio #2731: Rendering of Annotation Content Field on DataBrowser "unparses" the PV's (trace) EGU

ambarb, 2025-01-09:
Using cs-=studio version `4.6.1` on Red Hat Enterprise Linux release 8.10 (Ootpa)

I would expect  `Torr` to be displayed at the end of line 2 for the black trace called `XF:23ID1-VA{Dif-CCG:1}P-I`

[image]
[image]

kasemir, 2025-01-09:
Is this caused by having `..{..}..` braces in the PV name?

ambarb, 2025-01-09:
All PVs at NSLS-II have `..{..}..`.  Since I am only a user of cs-studio and EPCIS, this isn't something that is easy for me to test.  

This issue isn't critical to fix for me, but I do not know what other features may be impacted so I thought it is best to report it.

kasemir, 2025-01-09:
The formatting seems to work fine for us here where we don't have `..{..}..` in PV names, so I assume the problem is related to having them in there.

kasemir, 2025-01-10:
Well, I cannot reproduce. Created some PV with curly braces in the name, added an annotation with text at the end, and the value with units stays intact, not moving the units to the end

[image]

kasemir, 2025-01-10:
Hah, good news: I can reproduce!
Need to use a PV name like `ramp{Dif-CCG:1}def`. In there, `{..1}` is recognized as the placeholder `{1}`. The rest, `Dif-CCG:` is mistaken as some processing instructions, which gives some strange result because it's not a proper processing instruction, compare to the "advanced formatting options" in the dialog.

kasemir, 2025-01-10:
Thought that maybe a PV name with `{Dif-CCG:1}` where the `{..:1}` could be mistaken for a formatting option would make a difference, but still can't reproduce

kasemir, 2025-01-10:
Tried a dummy record with your exact example name, works fine:
```
record(calc, "XF:23ID1-VA{Dif-CCG:1}P-I")
{
  field(INPA, "XF:23ID1-VA{Dif-CCG:1}P-I")
  field(CALC, "A<10?A+1:0")
  field(EGU,  "units")
  field(SCAN, "1 second")
}
```